Resumer3d/info #1: Is it possible to port it to linux machine probably a raspberry pi?

amrithmmh, 2018-06-13:
It will be much useful if the software could be used in linux rather than just windows

Resumer3d, 2018-06-13:
Hi amrithmmh,  thanks for your good idea! we will implement this code to our new printer board's software that based on marlin(atmega2560).  if it is used in linux, there is also need some hardware like add a super capacitor.

amrithmmh, 2018-06-16:
@Resumer3d  really loved the idea .Hope you could make it opensource as this will provide much more support from the 3d printing community(marlin arduino are all opensource )
(i know you need to make a living but there are various successful opensource boards that are thriving ).
I didnt do much research when i first posted the question(sorry!), yes i understand that you have made both software and hardware for the Resumer3d which can be connected to any 3d printer.  I also noticed your kickstarter funding is below 50% of your targeted funding, do you intend to do this project even if the funding does not reach the goal?

hackaday3D, 2018-06-18:
Thanks! you encourage us to do it better! we will go on doing this project, now the shell mould is opened.

Resumer3d, 2018-06-18:
Thanks! you encourage us to do it better! we will go on doing this project, now the shell mould is opening.