Resumer3d/info #2: Mesh Bed Level

fsadak, 2018-09-06:
I know that the product does not support active bed level systems such as linear, bilinear or 3point because it is not possible to measure the level when there is a half job on the bed but there is no reason not to support the mesh level system which is a passive bed level system which can be used several times.

The M420 S1command activates the mesh bed level system and must be added to startup g-code section in the slicer app. But the Resumer 3D does not resend this command while resuming the remaining printing and the mesh level system does not work on the ongoing printing.

To solve this problem, a startup code system can be added which can be determined by the user. For example, like a similar command M5000 baud-rate (such as M5004 S1), the user can send to the Resumer 3D and have the mesh level system engaged in all future print resumes.

May be you can publish configuration wizard application with GUI.

hackaday3D, 2018-09-21:
Thanks @fsadak , your suggestions are very great. I am trying to support the auto bed level

blueeagle69, 2019-03-01:
Hi.
Is there any updates regarding auto bed levelling?
Thanks.