Resumer3d/info #7: Never connect to printer

Legired, 2019-01-07:
Hello,

I just buy a resumer 3d on amazon for my anycubic kossel printer

This printer use the same motherboard than Anycubic Mega i3

But, i can't make it work, i choose the right baudrate (250000) here my log of the Resumer Box Tool :

`connecting ......
start
echo:Marlin1.0.0
echo: Derniere MaJ: Dec 10 2018 20:08:28 | Auteur: (Leafy)
Compiled: Dec 10 2018
echo: Memoire libre: 3554 PlannerBufferBytes: 1232
echo:Stored settings retrieved
echo:Steps per unit:
echo:  M92 X80.00 Y80.00 Z80.00 E192.00
echo:Maximum feedrates (mm/s):
echo:  M203 X200.00 Y200.00 Z200.00 E200.00
echo:Maximum Acceleration (mm/s2):
echo:  M201 X2000 Y2000 Z2000 E2000
echo:Acceleration: S=acceleration, T=retract acceleration
echo:  M204 S2000.00 T2000.00
echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
echo:  M205 S0.00 T0.00 B20000 X20.00 Z20.00 E20.00
echo:Home offset (mm):
echo:  M206 X0.00 Y0.00 Z0.00
echo:Endstop adjustement (mm):
echo:  M666 X0.00 Y0.00 Z0.00
echo:PID settings:
echo:   M301 P22.20 I1.08 D114.00
G92 Z0.00
 probe_printer1
 probe_printer4
echo:Carte SD Ok
G92 Z0.00

 fail to connect 3D printer! wrong baud rate;
 reb? go to app boot to app
PA1 down
 Baud rate:250000;  Resumer3D:V1.1.4,compile date:Dec 26 2018 17:53:00
 Resumer3D:V1.1.4,compile date:Dec 26 2018 17:53:00
 en
USB reset
USB reset
CP2102 USB to UART Bridge Controller Select This Interface.
 Baud rate:250000;  Resumer3D:V1.1.4,compile date:Dec 26 2018 17:53:00`

When i connect the Resumer on my printer, he doesn't start, i have to put a external alimentation for make it work, but, he only flashing quickly, i try with other baudrate value, and same problem, i also change the cable for the resumer to the printer, and the cable for the alimentation, and same problem

i use the original firmware on my printer, is based on marlin V1

thx in advance

hackaday3D, 2019-01-15:
Hi Legired , sorry for reply late!    resumer3d detect the printer by sending gcode command  like "G92 z0.00" to printer. it seams that the printer did not return "ok", and the resumer3d will reboot after timeout.
can you send the gcode command  "G92 z0.00" to the printer by anyother tools and to see if it return "ok"?

Legired, 2019-01-15:
Hello,

When i send "G92 Z0.00" with repetier the printer return in the log "X0.00 Y0.00 Z0.00 E0.00 Count X42232 Y42232 X42232"

I try with marlin 1.1.19 and it's the same

hackaday3D, 2019-01-15:
no "ok" return?  in my printer with marlin1.1.9 ,it return like this
"X:0.00 Y:0.00 Z:0.00 E:0.00 Count X:0 Y:0 Z:0
ok"

Legired, 2019-01-15:
I have a "ok" after this line

hackaday3D, 2019-01-15:
it maybe the timeout in the resumer3d is short, could you try to connect the resumer3d to printer after the printer is started.

Legired, 2019-01-15:
When i plug the resumer3d during a print, the printer reboot and the print stop, the resumer 3d still flashing quickly

hackaday3D, 2019-01-17:
Hi, The v1.1.5beta.bin is online now, could you test it ?  thanks!

Legired, 2019-01-17:
Hello,

With the new firmware, i can connect sur resumer 3d to my printer.
 
But, when i turn off the printer during a print, when the resumer and the printer reboot, the resumer3d is blue and not red, so i can't resume the print

hackaday3D, 2019-01-17:
can you see the blue light flash slowly while printing?

hackaday3D, 2019-01-17:
the resumer3d can not save enough information while first layer printing , so make sure it have printed more than one layers.

Legired, 2019-01-17:
print at 50% and same problem ....

hackaday3D, 2019-01-17:
can you send me the log while printing?

Legired, 2019-01-17:
I connect the resumer3d on the printer and on my computer with the boxtool, open the port of the resumer 3d, but i don't have the log of the printer in the boxtool

hackaday3D, 2019-01-17:
There will be some log, if you unplug and plug the usb line to printer.

hackaday3D, 2019-01-17:
thanks, from the log "114_1:200001", it means the timeout after resumer3d send "M114" to printer. 
can you send me the link of your printer's control board? I will buy one to test.

Legired, 2019-01-17:
Here the link : 

https://fr.aliexpress.com/item/ANYCUBIC-Carte-M-re-3D-Imprimante-TriGorilla-carte-Principale-Compatible-Mega2560-RAMPS1-4-4-Couches-PCB/32846634194.html?spm=a2g0w.search0104.3.2.25795e84UFFYCj&transAbTest=ae803_3&ws_ab_test=searchweb0_0%2Csearchweb201602_1_10065_10068_319_10892_317_5735511_10696_453_10084_454_10083_10618_431_10304_10307_10820_10821_537_10302_536_5736111_5736011_10843_10059_10884_10887_100031_321_322_10103%2Csearchweb201603_52%2CppcSwitch_0&algo_pvid=7f862de2-0254-4dae-9708-5bdf2f17ce1e&algo_expid=7f862de2-0254-4dae-9708-5bdf2f17ce1e-0

Legired, 2019-02-27:
Hello, any news ?

Stranger74, 2019-04-25:
Hello, I'm having a similar problem with Alunar M508(mainboard HY_BASE_L V1.0 - ATMEGA 2560) and Marlin 1.1.9. The connection with 3D resumer seems impossible. I've upgraded 3D resumer to the beta 1.1.5 without success. Here my log from toolbox:

go to app boot to app
PA1 down
 Baud rate:250000;  Resumer3D:V1.1.5,compile date:Jan 17 2019 14:00:05
 Resumer3D:V1.1.5,compile date:Jan 17 2019 14:00:05
 en
USB reset
USB reset
 go to app boot to app
PA1 down
 Baud rate:250000;  Resumer3D:V1.1.5,compile date:Jan 17 2019 14:00:05
 Resumer3D:V1.1.5,compile date:Jan 17 2019 14:00:05
 en
USB reset
USB reset
 go to app boot to app
PA1 down
 Baud rate:250000;  Resumer3D:V1.1.5,compile date:Jan 17 2019 14:00:05
 Resumer3D:V1.1.5,compile date:Jan 17 2019 14:00:05
 en
USB reset
USB reset

Thank you