Resumer3d/info #8: OctoPrint

maxlinux2000, 2019-03-26:
Hello,
I use "octoprint" to print.
On my laptop I have CURA, connected wia wifi to octorpint in an orange pi pc, which is connected by usb to the printer (a clone of prusa i3 with gt2560 card).
I was thinking about putting the Resumer3D between the orange Pi and the printer, and putting an SD card in the printer.

In the manual I seem to understand that this can not be done. Is there any way to continue using the octoprint?

regards
MaX

hackaday3D, 2019-04-14:
hello,the latest resumer3D firmware support this case of putting resumer between octopi and printer. the resumer3D can pass through all the data between octopi and printer

cinealfa, 2019-12-10:
maybe you can update the manual?