Resumer3d/info #10: PA1 down

BlackienBad, 2019-04-05:
When I try to set the Baud rate I get this message:
 go to app boot to app
PA1 down
 Baud rate:115200;  Resumer3D:V1.1.3,compile date:Oct 30 2018 12:10:39
 Resumer3D:V1.1.3,compile date:Oct 30 2018 12:10:39
 en
and nothing happens

hackaday3D, 2019-04-14:
hello,it seems OK. after you set the baudrate the resumer3D will restart and display this information

BlackienBad, 2019-04-14:
> 
> 
> hello,it seems OK. after you set the baudrate the resumer3D will restart and display this information

I can't set the baudrate, this is the only message that I get, after this I tried to plug it in and when I pulled the charger nothing happened, I couldn't resume the print

hackaday3D, 2019-04-26:
is there some messages while the printer connect to resumer3D via USB.

kiroke, 2019-07-10:
I also have the same problem PA1 down so it doesn't work, even the firmware update has been tried but the result is the same, I sent an email for assistance I'm waiting for a reply .....

kiroke, 2019-07-11:
my resumer after having configured it (badly) I try to connect it to the printer, at the end of the work it turns on red ok, but if I try to restart the print it turns blue and then turns red.
Obviously it is caused by the message PA1 down, how to solve this problem I have to go back to the old UPS?

blueeagle69, 2019-07-24:
That's normal if the Resumer has saved gcode information. If you start another print it should go blue again. Or you can delete old info by holding down +- followed by OK. You're doing better than me. Since moving to an SKR 1.3 my Resumer will not connect it. And Eryone have been no help. Very disappointed considering I was a Kickstarter backer.