Resumer3d/info #12: Runout Sensor

blueeagle69, 2019-06-03:
Hi.
I have been testing my filament runout sensor made with a microswitch.
When the filament is inserted into the sensor the Resumer3D shows PA1 Down on BoxTool's output.
Is this correct, or should it say PA1 down when there is no filament in the sensor?
I just need to make sure I have my wiring to the microswitch correct.

Thanks!!

luojin012, 2019-06-06:
you are right, The PA1 pin is used for filament detect. it will output PA1 down when no filament.

blueeagle69, 2019-06-06:
Excellent,  thank you.
I will change the pin connections on my microswitch so the Resumet outputs
PA1 down when the filament is removed from the sensor.
Currently it outputs this when I insert filament!

Thanks for your help.

blueeagle69, 2019-06-08:
Just an update. Now I have changed the cable position on my microswitch, I tested the Resumer3D.
I pulled my short piece of filament I had inserted into my DIY sensor out, and shortly after the printer stopped and the nozzle moved away correctly :)
Thanks for your help.