Resumer3d/info #13: Bigtreetech SKR 1.3 Connection issue

blueeagle69, 2019-06-21:
Hi.

I have changed the baud rate of the Resumer3D to match that of the board, but despite that, it simply refuses to connect to my new board.
I am currently running the 1.1.4 firmware.
Is it worth trying the 1.1.5 beta?
Many thanks.

hackaday3D, 2019-11-21:
Hello blueeagle69, so sorry for delay replay, can you tell me the usb driver chip on the skr board? and use the toolbox to capture the log and send to me ?

blueeagle69, 2019-11-22:
Hi.
It appears it doesn't have a USB driver chip, it goes directly to the MCU.
https://m.facebook.com/groups/505736576548648?view=permalink&id=813987329056903

blueeagle69, 2019-11-22:
Do you mean the BoxTool?
Could you brush me up on collecting logs,  I've totally forgotten!

cinealfa, 2019-12-14:
any luck?